flipperdevices/flipperzero-good-faps #12: Picopass Rename Option

DarkEnder4583, 2023-08-08:
### Describe the enhancement you're suggesting.

I would like to request a “rename” option for the Picopass app. Currently, there is no simple way to rename any saved Picopass cards (outside of just manually changing it on a computer).

### Anything else?

(Y’all are great! Thanks for your work!)

skotopes, 2023-08-08:
@bettse I'll keep it here )

bettse, 2023-08-18:
@DarkEnder4583 Do you know how to build a fap off a branch?  If so, can you test https://github.com/flipperdevices/flipperzero-good-faps/pull/25 ?

DarkEnder4583, 2023-08-18:
Unfortunately, I do not. Sorry!

(As a side note, I've tried a couple times to build a fap, but for some reason can't run ./fbt commands on my shell, even though I've installed Git. If you had any tips or docs you could point me to, I would appreciate it.)

skotopes, 2023-08-18:
@DarkEnder4583 https://github.com/flipperdevices/flipperzero-ufbt then go into app folder and `ufbt launch`

bettse, 2023-08-18:
Code merged.  I've got some other Defcon-inspired features before we bump the version, but this will be in the part of it.

closed 2023-08-18 by bettse

DarkEnder4583, 2023-08-19:
@skotopes Thank you! That's very useful!

And thank you @bettse !