flipperdevices/flipperzero-good-faps #21: mass_storage: App frozen when ejected using Nautilus

dogtopus, 2023-08-12:
The app will freeze up when the disk is ejected from Nautilus (the GNOME file manager) under Linux. Though if I use the poweroff option in GNOME Disks or equivalent udisksctl commands it seems to work just fine. All of them go through udisks2 but seems like the shutdown sequence is slightly different for Nautilus.

Attached USB low-level [protocol capture](https://github.com/flipperdevices/flipperzero-good-faps/files/12327607/eject.zip) in PCAP format.

- Host OS: Fedora Workstation 38
- mass_storage version: 1.1
- Firmware version: 0.88.0

skotopes, 2023-08-16:
Fixed in latest update, soon will be release to catalog

dogtopus, 2023-09-11:
I can still reproduce this on latest release.

skotopes, 2023-09-11:
Dp you have debugger(stlink, wifiboard, jlink)?

dogtopus, 2023-09-11:
Yes I have a stlink clone. However I'm having some trouble running Flipper openocd (ABI incompatibility with Flatpak Freedesktop SDK) on my setup. I'll try to find some time to get it running on some other platform or using other openocd build and see what I can do.

skotopes, 2023-10-12:
Okay, looks like you've been testing wrong version. Ensure that you are running 1.2 from catalog, should be fixed there.

closed 2023-10-12 by skotopes