flipperdevices/flipperzero-good-faps #24: Bluetooth remote

Thepheonixmatt, 2023-08-16:
My Samsung s22 ultra will not connect to flipper unless I use the flipper app. When I try to select flipper in the Bluetooth settings it tells me an app is needed to use the app.

skotopes, 2023-08-17:
Did you open Remote app at that moment?

Thepheonixmatt, 2023-08-17:
After restarting the flipper and forgetting the 2 devices that flipper shows up as opening the app on flipper and pairing to the control it now works perfectly

skotopes, 2023-08-17:
@doomwastaken can you try to reproduce this issue?

doomwastaken, 2023-08-17:
@skotopes just tried this on flipper where I was previously connected, all works.  Forgot device and paired again, it works. Pairing to a "fresh device" works.  Tested on s23 ultra