flipperdevices/flipperzero-good-faps #30: dap link hang a dolphin with a large data stream on uart

Andrysky, 2023-08-23:
### Describe the bug.

Dolphin (dap link application hangs flipper) freezes if there is a lot of data on uart that the computer does not process (the uart port on the computer is not open)

### Reproduction

1. upload next code to esp32 board

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include "esp_log.h"

static const char *TAG = "UART TEST BUG REPORT";

static void echo_task(void *arg)
{
    while (1) {
            ESP_LOGI(TAG, "Hello flipper!, you have a bug()");
    }
}
void app_main(void)
{
    xTaskCreate(echo_task, "uart_echo_task", 2048, NULL, 10, NULL);
}
```

2. connect esp32 board to flipper zero (3v, gnd, 15(TX),16(RX)pins)

3. Open DAP Link(Apps->GPIO->DAP Link)
4. after 2 second press back button

5. result: the hourglass spins forever - it's impossible to get out, only rebooting by pressing the LEFT and BACK buttons helps.

### Target

Release 0.88.0

skotopes, 2023-08-23:
@Andrysky moved here ;-)

Skorpionm, 2023-10-02:
https://github.com/flipperdevices/flipperzero-good-faps/pull/51

fixed, but to check it, you need to add a PR to the main firmware that implements DMA reception in UART

closed 2023-10-10 by skotopes