flipperdevices/flipperzero-good-faps #33: Mass Storage consistently crashing when disk image is selected

brwyatt, 2023-09-03:
Oddly, it was working alright at first, or seemed to be, with a few random crashes here and there, but now it crashes every time. I've removed all disk images and even deleted and re-installed the application, but even creating a new image causes the device to crash after about 3 seconds. The error upon recovery is "MemManage".

Would be happy to provide any logs or debug information I can. I didn't see any log files that were obvious, and I'm not especially comfortable with Serial devices, but can poke around there if that's what's needed.

Mostly just logging this as Googling wasn't pulling anything up, and hopefully anyone else having the same issue might at least find this and know they aren't alone.

skotopes, 2023-09-03:
Which app are we talking about?

brwyatt, 2023-09-03:
In the title, the Mass Storage app: https://github.com/flipperdevices/flipperzero-good-faps/tree/dev/mass_storage

skotopes, 2023-09-03:
@brwyatt can you provide following:

- Flipper FW Version
- PC OS Version
- Steps to reproduce
- If you have serial adapter: using any terminal software you like connect to device console (GPIO13,14) and get the log at the moment of crash

brwyatt, 2023-09-03:
* FW version: Release 0.90.1
* PC OS version: crashes before I even plug it in... or was. I just tested trying to see if I plugged it in before it could crash what would happen, and it not only worked, but I can't seem to get it to crash. However, removing it from the Windows 11 computer, the Flipper Zero was locked up and needed a restart, but didn't get that behavior on the Ubuntu 18.04 computer.
* Steps to reproduce:
  1. Create a new image if one isn't created (any size)
  2. select it
  3. wait a couple seconds (unplugged from computer)
  4. Flipper crashes with a "MemManage" error.
* If I can get it to start crashing again, I will try this. It was happening consistently, not sure why it stopped. Even after replacing the SD card with a different one, and re-installing everything, it was still consistently crashing when any image was selected, but now it is not.

So not sure what was happening here, as it was happening on a completely clean/fresh SD card, too... consistently, but now it seems to be working. I'll poke around and see if I can "break" it again, and see about getting that serial logging going as well if I can.

For what it is worth, here's a theory I'm chasing down right now: this started happening after I kinda tricked it into having a >2GB image file (8GB, specifically, at least from the SD card's FS perspective, exFAT should be fine with this). But it was working for a bit (writing a Windows 11 install image to it, thought OS installers might be a good use for this kinda utility, but that needs >5GB). That would make sense if it was crashing on THAT image, but it did work at first, and then also wasn't working after deleting the file, or on other img files... and still wasn't working after a complete wipe and reinstall of the SD card. But I'll give it a try and see if trying that again triggers the same behavior... and then I'll see if I can figure out what caused it to start working again. I don't have high hopes for this theory, unfortunately.

skotopes, 2023-09-03:
We intentionally limited image size to 2GB to prevent issues with FatFS. Also read/write speed makes images of that size meaningless.

We'll check what we can do.

brwyatt, 2023-09-03:
Just created a new image (after the previous 2 I was already testing with) and it immediately crashed when unplugged from the computer. Tried plugging it in first and then selecting the image, and so far so good.

brwyatt, 2023-09-03:
Yeah, this was crashing on even 4MB (and smaller), even with no "large images" present (or ever present on that SD card install). The one it just crashed on was 2GB.

skotopes, 2023-09-03:
Clearly something unusual happens on your device, I can not reproduce it. Are you sure that your SD filesystem is not damaged?

brwyatt, 2023-09-03:
Tried two SD cards (for disclosure, the first was 512GB, the second (current) is 64). The first was originally used in a Steam Deck, and reformatted by the Flipper Zero before being used, and the second used to be in a Nintendo Switch and was also formatted by the Flipper Zero before having the firmware freshly upgraded onto it and everything re-installed (but did not factory reset the device itself).

I've also just poked at it today, and can continue to reproduce the behavior - if I select an image and do NOT plug it into a computer's USB, it crashes the Flipper Zero after about 3 seconds. If I DO plug it into a computer, it seems to work as expected.

This behavior also makes me think it'll be tricky (impossible?) to get logs from the device using a USB serial connection, since the crash only happens when it is NOT plugged into USB.

Made a video demonstrating the issue: https://photos.app.goo.gl/8szsREGv3SVuFyjg9

When it was plugged into the laptop, the system (Ubuntu 20.04 in this case) recognized the storage device.

brwyatt, 2023-09-03:
FWIW, I think this can be super low priority here, just figured it was worth having something somewhere in case someone else runs into this (or something similar) so something comes up in a search. Especially given there's a workaround ("have the device plugged in BEFORE selecting an image") that seems to mitigate the impact.

skotopes, 2023-09-04:
Can not reproduce it. We need to debug it on your side. Do you have any debugger that we support(second flipper, wifi devboard, st-link, jlink)?

brwyatt, 2023-09-04:
I do actually have a WiFi devboard, with Marauder firmware at present. Might be able to find time tomorrow (though no guarantee, unfortunately).

Let me know if there's another firmware for the devboard you'd prefer I use and I can flash that instead of what's on there now.

skotopes, 2023-09-04:
Ensure that you have esptool installed in your path, then clone firmware repository and do `./fbt devboard_flash` to flash blackmagic into wifi board.

Then do following:
-`git co 0.90.1`
- `./fbt blackmagic`
- In GDB shell `continue`
- Start mass_storage app
- One more time `continue` in gdb shell
- In App: open file, wait for crash
- In gdb shell: `thread apply all bt`
- Copy output to file and post here

Feel free to ping me on discord if you need help:`@skotopes`

brwyatt, 2023-09-05:
Didn't get a chance today to try the real debugging capture (I started, but was running into GLIBC version issues on Ubuntu 18.04, will try on 20.04 tomorrow).

BUT I did notice something idly poking at it:

If I have no USB cable plugged in, it won't crash. If I have a USB cable plugged into the Flipper Zero (the one it came with, specifically), but not a computer, it seems to crash.

Going to try and get my friends from earlier to try with that information and see if they can replicate it, and I'm going to see if I can get the debugging stuff to work tomorrow on Ubuntu 20.04 (if not, I'll see if I can get it working with WSL on Windows)

brwyatt, 2023-09-05:
Oddly, I can't get it to act the same with the debugger. Instead of crashing, it seems to lock up instead (back button doesn't respond). Killing the debugger results in a HardFault error on the device.

The same process in a non-crashing configuration, I'm able to use the back button as normal/expected, it's only when it is configured in the setup that causes the crash that it locks up.

I'm not sure how helpful this will actually be, but this is the output from the lockup condition (so the setup where it normally crashes, but just locks up when the debugger is attached):

```
$ ./fbt blackmagic
	SDKCHK	firmware/targets/f7/api_symbols.csv
API version 35.1 is up to date
[...]/flipperzero-firmware/toolchain/x86_64-linux/bin/arm-none-eabi-gdb-py -ex "source ./scripts/debug/gdbinit" -ex "target extended-remote /dev/ttyACM0" -q -ex "monitor swdp_scan" -ex "monitor debug_bmp enable" -ex "attach 1" -ex "set mem inaccessible-by-default off" -ex "source ./scripts/debug/FreeRTOS/FreeRTOS.py" -ex "source ./scripts/debug/flipperapps.py" -ex "source ./scripts/debug/flipperversion.py" -ex "fap-set-debug-elf-root build/f7-firmware-D/.extapps" -ex "source ./scripts/debug/PyCortexMDebug/PyCortexMDebug.py" -ex "svd_load ./scripts/debug/STM32WB55_CM4.svd" -ex compare-sections -ex fw-version build/f7-firmware-D/firmware.elf
Reading symbols from build/f7-firmware-D/firmware.elf...
Remote debugging using /dev/ttyACM0
LL_LPM_EnableSleep () at lib/stm32wb_hal/Inc/stm32wbxx_ll_cortex.h:314
314	  CLEAR_BIT(SCB->SCR, ((uint32_t)SCB_SCR_SLEEPDEEP_Msk));
You are now detached from the previous target.
Available Targets:
No. Att Driver
 1      STM32WBxx (secure) M4
Target does not support this command.
Attaching to program: [...]/flipperzero-firmware/build/f7-firmware-D/firmware.elf, Remote target
LL_LPM_EnableSleep () at lib/stm32wb_hal/Inc/stm32wbxx_ll_cortex.h:314
314	  CLEAR_BIT(SCB->SCR, ((uint32_t)SCB_SCR_SLEEPDEEP_Msk));
Support for Flipper external apps debug is loaded
Set 'build/f7-firmware-D/.extapps' as debug info lookup path for Flipper external apps
Attaching to Flipper firmware
Loading SVD file ./scripts/debug/STM32WB55_CM4.svd...
Section .isr_vector, range 0x8000000 -- 0x800013c: MIS-MATCHED!
Section .text, range 0x8000140 -- 0x8087664: MIS-MATCHED!
Section .rodata, range 0x8087668 -- 0x80ab350: MIS-MATCHED!
Section .ARM, range 0x80ab350 -- 0x80ab358: MIS-MATCHED!
Section .init_array, range 0x80ab358 -- 0x80ab35c: MIS-MATCHED!
Section .fini_array, range 0x80ab35c -- 0x80ab360: MIS-MATCHED!
Section .data, range 0x80ab360 -- 0x80ab980: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file

Firmware version on attached Flipper:
	Version:     0.90.1
	Built on:    31-08-2023
	Git branch:  0.90.1
	Git commit:  384e950a
	Dirty:       False
	HW Target:   7
	Origin:      Official
	Git origin:  https://github.com/flipperdevices/flipperzero-firmware
(gdb) continue
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08073d10 in canvas_draw_u8g2_bitmap (u8g2=0x0, x=<optimized out>, y=<optimized out>, w=<optimized out>, h=176 '\260', bitmap=0x20007478 "\240t", rotation=204) at applications/services/gui/canvas.c:339
339	    switch(rotation) {
New application loaded. Adding debug info
Loading debug information from build/f7-firmware-D/.extapps/mass_storage_d.elf
add symbol table from file "build/f7-firmware-D/.extapps/mass_storage_d.elf" at
	.text_addr = 0x2000f584
	.bss_addr = 0x2000cf9c
	.rodata_addr = 0x2000d284
Failed to execute GDB command 'add-symbol-file -readnow build/f7-firmware-D/.extapps/mass_storage_d.elf 0x2000f584 -s .bss 0x2000cf9c -s .rodata 0x2000d284': build/f7-firmware-D/.extapps/mass_storage_d.elf: No such file or directory.
Failed to load debug info for AppState(name='Mass Storage', text_address=536933764, entry_address=536941245, other_sections={'.bss': 536924060, '.rodata': 536924804}, debug_link_elf='mass_storage_d.elf', debug_link_crc=1173619588)
(gdb) continue
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
__furi_crash () at furi/core/check.c:140
140	        __furi_check_message = "furi_assert failed";
New application loaded. Adding debug info
Loading debug information from build/f7-firmware-D/.extapps/mass_storage_d.elf
add symbol table from file "build/f7-firmware-D/.extapps/mass_storage_d.elf" at
	.text_addr = 0x2000f584
[... 7 lines cut ...]

^C
Program received signal SIGINT, Interrupt.
__furi_crash () at furi/core/check.c:141
141	    } else if(__furi_check_message == (void*)__FURI_CHECK_MESSAGE_FLAG) {
New application loaded. Adding debug info
Loading debug information from build/f7-firmware-D/.extapps/mass_storage_d.elf
add symbol table from file "build/f7-firmware-D/.extapps/mass_storage_d.elf" at
	.text_addr = 0x2000f584
	.bss_addr = 0x2000cf9c
	.rodata_addr = 0x2000d284
Failed to execute GDB command 'add-symbol-file -readnow build/f7-firmware-D/.extapps/mass_storage_d.elf 0x2000f584 -s .bss 0x2000cf9c -s .rodata 0x2000d284': build/f7-firmware-D/.extapps/mass_storage_d.elf: No such file or directory.
Failed to load debug info for AppState(name='Mass Storage', text_address=536933764, entry_address=536941245, other_sections={'.bss': 536924060, '.rodata': 536924804}, debug_link_elf='mass_storage_d.elf', debug_link_crc=1173619588)
(gdb) Quit
(gdb) quit
Detaching from program: [...]/flipperzero-firmware/build/f7-firmware-D/firmware.elf, Remote target
Failed to set debug mode: Cannot access memory at address 0x200008cc
[Inferior 1 (Remote target) detached]
scons: *** [phony_blackmagic] Build interrupted.

********** FBT ERRORS **********
phony_blackmagic: Build interrupted.
```

I might be able to try some other things, though

brwyatt, 2023-09-05:
On another note, I was able to get one confirmation from someone else that they could get similar behavior when a USB cable was plugged in to the Flipper Zero side only.

mmiller7, 2023-09-05:
Verified on mine.

Flipper unplugged (no cable connected) = works
Flipper plugged into computer = works
**Flipper with USB-A to USB-C cable (computer end unplugged) = CRASH
Flipper with USB-A to USB-C cable (plugged into wall charger) = CRASH**
Flipper with USB-C to USB-C2.0 cable (computer end unplugged) = works
Flipper with USB-C to USB-C2.0 cable (plugged into wall charger) = works

dogtopus, 2023-09-09:
See #20 and #21. Should be fixed on main but hasn't been deployed yet.

skotopes, 2023-09-10:
@dogtopus nope, 1.1 includes those fixes

skotopes, 2023-09-10:
But, yes. @brwyatt which app version do you use?

brwyatt, 2023-09-10:
Not entirely sure. Must be before 1.1, as that's what the app currently says is available for install, but the Flipper isn't staying connected to my phone long enough for me to be sure, it might be the installed version (I would have installed it in the last week or two, though).

I'll keep trying to verify, if I can get Bluetooth to cooperate.

brwyatt, 2023-09-10:
Bluetooth finally cooperated, appears to be 1.1.

skotopes, 2023-09-10:
As for blackmagic log you showed: you need to enable debug in system setting

brwyatt, 2023-09-10:
Yeah, I did, or I thought I did. It wouldn't connect at all without that being enabled.

But, unfortunately, the behavior appears to be different when the debugger is attached (freezes instead of crashing, and then exiting the debugger causes a different crash). I won't be able to try the debugger again until Monday, unfortunately.

Were you able to reproduce the issue on your end with the added details we found where it only happens when the USB cable is plugged into the Flipper Zero but NOT a computer? Seems to not happen if no USB cable is installed at all, or if the cable is plugged into the computer on the other end.

skotopes, 2023-09-10:
No, you are the only one who can reproduce this issue

dogtopus, 2023-09-10:
@skotopes Looks like I can still reproduce #21. Should I reopen that issue?

brwyatt, 2023-09-10:
Was able to get some friends who previously couldn't reproduce the issue to be able to with the new details. (see @mmiller7 's comment)... Was really hoping it was more universal after that... dang.

Alright, I'll give the debugger another shot tomorrow (if I don't forget), see if I can get it to actually work right.

skotopes, 2023-09-11:
> @skotopes Looks like I can still reproduce #21. Should I reopen that issue?

Yes. Please attach gdb backtrace If you have debugger(or wifi with blackmagic).

dogtopus, 2023-10-10:
Sorry. Still haven't got time to look into it.

skotopes, 2023-10-12:
Okay, looks like you've been testing wrong version. Ensure that you are running 1.2 from catalog, should be fixed there.

closed 2023-10-12 by skotopes

mmiller7, 2023-10-22:
> Ensure that you are running 1.2 from catalog

Is there a way to double check the version installed on the Flipper?  I'm not sure what you mean by "catalog".

skotopes, 2023-10-30:
@mmiller7 https://lab.flipper.net/apps/mass_storage