flipperdevices/flipperzero-good-faps #43: mfkey32 error while saving keys to file mf_classic_dict_user.nfc

jfima, 2023-09-16:
i push the button GIVE ME THE KEYS
Attacking nonce 1 of 5

after that i get this content inside mf_classic_dict_user.nfc file

Error: mfkey run killed on timeout
c58380f171a8

all previous keys were destroyed

jfima, 2023-09-16:
To be honest, now I don’t understand where it is better to write to this repository or to the repository of the script author.

skotopes, 2023-09-16:
Which script are we talking about?

jfima, 2023-09-16:
> Which script are we talking about?

I guess the script is from this place /flipperdevices/flipperzero-good-faps/mfkey32

skotopes, 2023-09-16:
It's script, it is app. Here is the right place.

jfima, 2023-09-16:
I can send you a .mfkey32.log file that you can use by placing it in the /ext/nfc folder to reproduce this error. 

**Sec 0 key A cuid 2a234f80 nt0 5791a2d9 nr0 2b009acf ar0 ed570fc9 nt1 946e9749 nr1 d3f32722 ar1 bf727112
Sec 14 key A cuid 2a234f80 nt0 7fb7f403 nr0 22edfb97 ar0 f1a884ef nt1 62895d1c nr1 934e9592 ar1 8caf8b0e
Sec 0 key A cuid 2a234f80 nt0 4396a927 nr0 e40df916 ar0 c1c2f941 nt1 723b1be1 nr1 73c2a36c ar1 b5640c8b
Sec 0 key A cuid 2a234f80 nt0 d348bb83 nr0 84a5958d ar0 6d30dbce nt1 36891434 nr1 17d4eabb ar1 9ee6bd21
Sec 0 key A cuid 2a234f80 nt0 9840a9d8 nr0 7b1e8c18 ar0 c7957482 nt1 c75940ec nr1 eb2c794d ar1 bc6f31c8**

It will write this data 

**Error: mfkey run killed on timeout
c58380f171a8**

to the mf_classic_dict_user.nfc if you push button GIVE ME THE KEYS https://lab.flipper.net/nfc-tools

i have latest Firmware 0.91.4 and latest mfkey32 app

noproto, 2023-09-16:
Hello @jfima !

This is not an error in the Flipper Zero Mfkey32 application, you are using https://lab.flipper.net/ (the [error](https://github.com/flipperdevices/lab.flipper.net/blob/c1c609190e68662ab66af0272a0951473a8bbc65/frontend/src/util/mfkey32v2/mfkey.js#L5) you reference is only returned by lab.flipper.net). lab.flipper.net has multiple reported issues with timeouts and writing those strings to the user dictionary. Your issue would be best submitted to https://github.com/flipperdevices/lab.flipper.net.

I ran your nonces on the Mfkey32 Flipper app (which has no issues to the best of my knowledge) and it identified one key: `C58380F171A8`.

[image]

To ensure that the Mfkey32 Flipper app did not have any missed coverage, I also ran the nonces in [mfkey32v2](https://github.com/equipter/mfkey32v2) which returned the same key:

```bash
$ cat crack.sh
#!/bin/bash
./mfkey32v2 2a234f80 5791a2d9 2b009acf ed570fc9 946e9749 d3f32722 bf727112
./mfkey32v2 2a234f80 7fb7f403 22edfb97 f1a884ef 62895d1c 934e9592 8caf8b0e
./mfkey32v2 2a234f80 4396a927 e40df916 c1c2f941 723b1be1 73c2a36c b5640c8b
./mfkey32v2 2a234f80 d348bb83 84a5958d 6d30dbce 36891434 17d4eabb 9ee6bd21
./mfkey32v2 2a234f80 9840a9d8 7b1e8c18 c7957482 c75940ec eb2c794d bc6f31c8
$ ./crack.sh |grep 'Found Key'
Found Key: [c58380f171a8]
```

So the issue you are experiencing with this Mfkey32 method is isolated to lab.flipper.net. For more information on the available Mfkey32 methods, see: https://docs.flipper.net/nfc/mfkey32.

Slipn3r, 2023-09-25:
Fixed, sorry for inconvenience. To clear the garbage from your user dictionary just run mfkey again (press that 'give me the keys' button).

closed 2023-09-25 by Slipn3r

noproto, 2023-09-25:
> Fixed, sorry for inconvenience. To clear the garbage from your user dictionary just run mfkey again (press that 'give me the keys' button).

It's good that you fixed your issue @Slipn3r . However note that it is still inapplicable to this application and does not resolve the issue with the web version of mfkey32v2.

Slipn3r, 2023-09-25:
https://github.com/flipperdevices/lab.flipper.net/commit/800848185319dd8d8b86c775dc510c53db1de7e9 should (and did according to my research) fix the problem. Could you elaborate on why it doesn't?

noproto, 2023-09-25:
> [flipperdevices/lab.flipper.net@8008481](https://github.com/flipperdevices/lab.flipper.net/commit/800848185319dd8d8b86c775dc510c53db1de7e9) should (and did according to my research) fix the problem. Could you elaborate on why it doesn't?

Sorry. I did not see that commit, and I thought @jfima was following up saying they fixed it by removing the error from their dictionary file (it's still early here). Thank you for getting it fixed on the lab.flipper.net implementation @Slipn3r .