flipperdevices/flipperzero-good-faps #58: Add Davis Instruments to Weather Station

SolarSciencePup, 2023-10-02:


skotopes, 2023-10-02:
Pleas provide:
- Exact model and vendor name
- Raw recordings of it's signal at different temperatures

Skorpionm, 2023-10-02:
photo of the remote module, photo of the receiving module, raw recordings of transmissions at increasing temperatures (when only 1 decimal place changes >4 pieces, records with changes in full degrees at positive and negative temperatures >10 pieces) with the obligatory photo that shows the receiving module.  recordings and photos of the receiving module, when setting up an external module on different channels, if more than 1 module is supported.  raw recording of readings when removing and inserting batteries, at the same temperature

SolarSciencePup, 2023-10-03:
Removing and reinserting batteries I can't get

SolarSciencePup, 2023-10-03:
Do have FCC ID though

SolarSciencePup, 2023-10-03:
> Pleas provide:
> - Exact model and vendor name
> - Raw recordings of it's signal at different temperatures

Vantage Vue Wireless Integrated Sensor Suite Weather Station - SKU 6357, 6357M

By Davis Instruments

skotopes, 2023-10-09:
FCC ID says that this device uses FSK in a range between 902 - 927 MHz with channelization scheme.
It sounds slightly more complex than common weather stations and requires different architecture this sensor.

At this point we are not planning to reverse engineer it, but maybe someone from community will do.
Also because of the radio protocol complexity the best option will be to move this sensor to separate application.

closed 2023-10-09 by skotopes