flipperdevices/flipperzero-good-faps #65: Picopass incorrectly emulating configuration cards

Antiklesys, 2023-10-24:
When trying to emulate a configuration card in personalization mode, the Flipper Zero seems to emulate it incorrectly. I have tried reading it with a Proxmark3 and the following output is shown. It seems the Flipper is somehow trying to emulate blocks 32 to 255 only.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The configuration card I'm trying to emulate is attached. This works perfectly on the PM3 as well as on a card in personalization mode. 
[Config_Reset_Standard.picopass.txt](https://github.com/flipperdevices/flipperzero-good-faps/files/13113686/Config_Reset_Standard.picopass.txt)

From the flipper zero logs I couldn't seem to find anything special:

[image]

nvx, 2023-10-24:
other way around, it looks like the PM3 is trying to read blocks above 31, and the flipper isn't responding (which seems correct considering it's a 2k card).

Antiklesys, 2023-10-24:
But why would it? The picopass is emulating the dump of an actual card. If the issue was on the PM3 side it would have displayed the same behavior on the actual card to my understanding. I know picopass in the past had issues when the key was just 0s, I'm wondering here if the issue is due many of the blocks in the dump being 0s.

nvx, 2023-10-24:
Interesting, I thought you had constructed the dump by hand. There must be some difference in emulation that's triggering the PM3 to try and dump those blocks when it's not with the real card, or perhaps it always tries that but a real card responds differently vs the flipper for invalid block access.

Can you send me the PM3 trace file of doing the same command against the real card vs the flipper via Discord?

Antiklesys, 2023-10-24:
Yes I did construct the dump by hand on the Flipper but based on the dump from the card itself from the PM3, as the flipper refuses to read/dump a picopass card with "invalid data" to its saying (data is valid, just not actual credentials but instead is a config card).

Yes I'll pm you for the other bits.

bettse, 2023-12-22:
config cards aren't considered a core supported feature, so for now I'm closing this.

closed 2023-12-22 by bettse

nvx, 2023-12-24:
oh yeah I forgot to reply to this, the issue is with the dump, when you run `hf iclass sim -t 3` it overrides the config block to be an iclass 2k tag using a hardcoded value - not the value in the dump file, while the flipper emulation correctly uses the dump file config block. Your config block identifies as a `[+]     Card type.... PicoPass 32K with current book 16K / 2` according to `hf iclass info` which explains with the proxmark3 is attempting to dump more than the 2k blocks.

For reference your config block is `FF FF FF FF F9 FF FF BC`, the config block the proxmark3 codebase uses is `12 FF FF FF 7F 1F FF 3C 00 00`. Changing the config block in your dump to the PM3 one will replicate the behaviour of the proxmark3 simulation code.