flipperdevices/flipperzero-good-faps #99: missing support for Acurite 5n1 and lightning detector (?)

godmar, 2023-12-31:
When I run `rtl_433` with a [NooElec device](https://www.nooelec.com/store/sdr/sdr-receivers/nesdr-mini.html) it's picking up various sensors at my location. OTOH, F0 picks up only 2 of them.

F0 picks up this one:
```
model     : Acurite-Tower id        : 10450
channel   : B            Battery   : 1             Temperature: 21.0 C       Humidity  : 45 %          Integrity : CHECKSUM
```
and this one:
```
model     : Acurite-Tower id        : 9064
channel   : B            Battery   : 1             Temperature: 8.7 C        Humidity  : 50 %          Integrity : CHECKSUM
```

which I determined based on the `id` printed on F0's interface.

F0 misses the following stations for which I'm showing rtl_433's output:
```
model     : Acurite-6045M id        : 15821
channel   : A            Battery   : 1             temperature: 49.6 F       humidity  : 47 %          strike_count: 226         storm_distance: 17        active_mode: 0
rfi_detect: 0            data_exception: 0         raw_message: fdcd6faf0fb8711131
```
and
```
model     : Acurite-5n1  message_type: 56          id        : 3093
channel   : A            sequence_num: 2           Battery   : 1             wind_speed: 0.0 km/h      temperature: 44.4 F       humidity  : 60 %          Integrity : CHECKSUM
```
which I believe to be [Acurite 5-in-1 devices and lightning detector](https://www.acurite.com/shop-all/weather-instruments/weather-stations/acurite-iris-personal-weather-stations/hd-display-with-5-in-1-weather-station-and-lightning-detector.html). 

On an unrelated note, does the modulation (AM270, AM650, etc.) setting in the configuration matter?  I seem to be able to pick up the weather station it does pick up no matter what the setting is.

skotopes, 2024-01-02:
@godmar we need some raw recording for it

RedChops, 2024-01-26:
I can get you a recording for the 5n1, I haven't had my lightning detector hooked up in ages due to strange interference.

Is there a format or sample that I can look at to get you the right thing?

godmar, 2024-01-26:
> I can get you a recording for the 5n1, I haven't had my lightning detector hooked up in ages due to strange interference.

Do these things even work?  Mine also constantly detects "interference" no matter where I put it. I had them send me a replacement unit, and the replacement unit does the same.

RedChops, 2024-01-26:
> > I can get you a recording for the 5n1, I haven't had my lightning detector hooked up in ages due to strange interference.
> 
> Do these things even work? Mine also constantly detects "interference" no matter where I put it. I had them send me a replacement unit, and the replacement unit does the same.

Mine worked at my last house out in the country, but since moving to the city I think I'm getting interference from electrified rail. They're sensitive I guess