flipperdevices/flipperzero-good-faps #100: Missing support for "Fine Offset Electronics WH24, WH65B, HP1000 & derivatives

dclaar, 2024-01-01:
I'd like to see support for this protocol. 

I know that this is open source, and I'm happy to contribute, but it's a little difficult figuring out how to start. Not the github stuff, or the tool chain, but more the structure of the files, what's a "SubGhzBlockConst", etc. I'm trying to wrap my head around the protocol files--and on translating rtl_433 to flipper--but I haven't found a lot of documentation yet.

I have an Ambient Weather 2902, and the sending unit identifies as "fineoffset-WH24" in rtl_433:
```
time      : 2023-12-29 19:30:32
model     : Fineoffset-WH24                        ID        : 49
Battery   : 1            Temperature: 13.2 C       Humidity  : 93 %
Wind direction: 43       Wind speed: 0.0 m/s       Gust speed: 0.0 m/s
Rainfall  : 2809.5 mm    UV        : 2             UVI       : 0
Light     : 0.0 lux      Integrity : CRC
```
It's at 915mhz. It's different than the current ambient_weather.c:
```
 * This decoder handles the 433mhz/868mhz thermo-hygrometers.
 * The 915mhz (WH*) family of devices use different modulation/encoding.
 ```
This protocol was [added to rtl_433 in 2018](https://github.com/merbanan/rtl_433/pull/845/commits/bbe603f70758cb29cacb24cfed459804ece246ee), discussion of the signal is [here](https://github.com/merbanan/rtl_433/issues/844).

The format is described as: "The sensor sends a package each ~16 s with a width of ~11 ms. The bits are PCM modulated with Frequency Shift Keying"

Thanks!

skotopes, 2024-01-02:
@dclaar we need some RAW records first

skotopes, 2024-01-03:
RAW files you recorded are useless. Looks like you were using incorrect modulation settings: rtl433 is suggesting FSK

dclaar, 2024-01-03:
Ah, OK, I don't see an option for FSK.  Maybe one of the FM settings?

This was kind of why I was hoping that someone had written a doc on "this is how you do this" so I wouldn't waste people's time.  Here's one with each FM setting.
[RAW-20240102-192241.sub.txt](https://github.com/flipperdevices/flipperzero-good-faps/files/13815456/RAW-20240102-192241.sub.txt)
[RAW-20240102-192149.sub.txt](https://github.com/flipperdevices/flipperzero-good-faps/files/13815455/RAW-20240102-192149.sub.txt)

skotopes, 2024-01-03:
yes FSK is subtype of FM

skotopes, 2024-01-03:
Looks like default config is not going to work or this record was done incorrectly.

Does rtl433 shows you anything when you replay it from flipper?

dclaar, 2024-01-20:
I'm stymied and frustrated. When I put the flipper right next to the transmitter, the RSSI indicator doesn't even twitch. There's no indication that it is receiving anything. When I replay the recording, nothing happens. But the unit is clearly transmitting: rtl-433 sees it. So I don't know where to go from here.

ravenium, 2024-06-16:
Just found this topic - I've got a WS-5000 I can try some captures on.