flipperdevices/flipperzero-good-faps #103: Support for Kedsum-TH weather stations

s1shed, 2024-01-02:
### Describe the enhancement you're suggesting.

Add support for Kedsum-TH weather stations. This is supported by rtl_433 (https://github.com/merbanan/rtl_433/blob/master/src/devices/kedsum.c)

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```
time      : 2023-12-22 22:44:40
model     : Kedsum-TH    ID        : 158
Channel   : 3            Battery level: 0.000      Flags2    : 0             Temperature: -6.00 C      Humidity  : 89 %          Integrity : CRC
*** Saving signal to file g001_433.92M_250k.cu8 (296332 samples, 655360 bytes)
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```
time      : 2023-12-22 22:44:42
model     : Kedsum-TH    ID        : 26
Channel   : 1            Battery level: 0.000      Flags2    : 0             Temperature: 19.61 C      Humidity  : 33 %          Integrity : CRC
*** Saving signal to file g002_433.92M_250k.cu8 (503792 samples, 1048576 bytes)
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```
time      : 2023-12-22 22:45:09
model     : Kedsum-TH    ID        : 250
Channel   : 2            Battery level: 0.000      Flags2    : 0             Temperature: 20.22 C      Humidity  : 37 %          Integrity : CRC
*** Saving signal to file g003_433.92M_250k.cu8 (299724 samples, 655360 bytes)
```

The file [kedsum-th.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13756701/kedsum-th.zip) contains the .cu8 files. 

I converted these to sub files using EvilPete's `subghz_ook_to_sub.py`.  Only one ([g002_433.92M_250k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/13756710/g002_433.92M_250k.zip)) will be detected by rtl_433 when transmitted with the FlipperZero, so I did not include the others.

Please let me know if I can provide anything else to help get this device supported.

s1shed, 2024-01-04:
Hi, sorry for the delay.

Attached is a zip file ([kedsum-weather-th_raw.zip](https://github.com/flipperdevices/flipperzero-good-faps/files/13834659/kedsum-weather-th_raw.zip)) with several RAW signals captured with the Flipper, both positive and negative tempertures and various humidity levels.

All of these captures are with a station transmitting on channel 1 of 3. The files within are named with the following format: `degC(humid%).sub`. 

Please let me know if anything else would be of use for me to provide.

skotopes, 2024-01-05:
@Skorpionm FYI

Skorpionm, 2024-01-06:
Chek https://github.com/flipperdevices/flipperzero-good-faps/pull/104
you did not send more than one file with different transmitter IDs (it changes when you take out and insert the batteries), all files are recorded on channel 1, all files have a good battery charge. and it is not clear whether the transmitter has a button for forced data sending, there is no photo either (....
I did what I think is correct, but please check it on different channels, with different IDs and other cases.

s1shed, 2024-01-06:
> Chek #104 you did not send more than one file with different transmitter IDs (it changes when you take out and insert the batteries), all files are recorded on channel 1, all files have a good [...]

Thanks for the quick work! I'm sorry that my submission wasn't as helpful as it could have been. 

My FZ can receive signals from one of the weather stations that is currently close, at least. It's on channel 3.

[image]

I'll do some tests with it on different channels and with different IDs and will report my results.

s1shed, 2024-01-06:
It looks good from my end. :+1:  I removed the batteries a few times and switched the channels. The data received by the FZ matches what's on the display.

[image]
[image]

s1shed, 2024-01-06:
> Chek #104 you did not send more than one file with different transmitter IDs (it changes when you take out and insert the batteries), all files are recorded on channel 1, all files have a good [...]

The vendor provides the following photos (among others):

[image]
[image]

The transmitters only have two controls: a button to switch the display from °C to °F and a switch to change the channel.

s1shed, 2024-01-06:
I'll continue testing but by all indications this looks good so far.

[image]

Skorpionm, 2024-01-07:
The only question is, is the battery charge displayed correctly?  everything else seems similar

s1shed, 2024-01-07:
> The only question is, is the battery charge displayed correctly? everything else seems similar

The units themselves do not show the battery charge, but the values in the Weather Station fap match what `rtl_433` receives. 

[image]
[image]

[image]
[image]

I think the battery charge values can be considered to be correct.

Skorpionm, 2024-01-08:
Ok!

skotopes, 2024-01-08:
@Skorpionm merged, please push update to catalog

closed 2024-01-08 by skotopes