flipperdevices/flipperzero-good-faps #112: version 1.3 OF spi mem manager: XM25QH64C shows up as "Unknown SPI Chip"

devicemodder, 2024-01-23:
Accidentally closed the last post, sorry...

Anyways, in the changelog for SPI MEM MANAGER V.1.3, it shows that XM25QH64C support was added. i connected one of those chips and it's not being detected.

mine though is made by XMC, but has the same part number. Is there any way to get this to work? I have a device with this chip that i need to reflash the firmware on.

skotopes, 2024-01-23:
- Check wiring (maybe try to connect another chip)
- Check part number on the chip
- Check if anything detected or not in logs
- Is it new chip or old one? It may be damaged, try another one.

devicemodder, 2024-01-23:
Double checked wiring, its all correct, and the chip is known good, as I
pulled it from a working IP camera, just need to dump it and get the admin
password as the cam doesn't have a reset button.

I'll try another chip, and where on the flipper, might I find the logs?

skotopes, 2024-01-23:
Logs are available in CLI or on GPIO UART.

https://docs.flipper.net/development/cli#_yZ2E
or
https://docs.flipper.net/development/hardware/wifi-developer-board/reading-logs

devicemodder, 2024-01-23:
Tried another chip, slightly different part no... it reads fine. and i don't see the XM25QH64C in the flipper chip list. just the XM25L chips.

devicemodder, 2024-01-23:
Update: Very odd... for some reason, the flipper app store is installing version 1.0, even though version 1.3 is listed in the store.