flipperdevices/flipperzero-good-faps #147: weather station has way weaker RX

DavidIgor, 2024-02-17:
Im using latest unleashed and weatherstation app is just not pickng signals unless Im like 1meter away from sensor!
this has been a thing for few months now so several unleashed updates.... 
however I remember app working perfectly - wallk on street would pick up few different sensors from like 20-50 meters.

please let me know how to fix it . 
thanks

giddyhup, 2024-03-11:
Just to let @DavidIgor know that he is not alone: I am observing the same behavior.

skotopes, 2024-03-11:
You guys do understand that we don't support 3rd party firmware?

closed 2024-03-11 by skotopes

giddyhup, 2024-03-11:
@skotopes yes, I understand that and appreciate the reply. For the sake of clarity, are you implying that 3rd party firmware breaks the Weather Station app?

skotopes, 2024-03-11:
@giddyhup no bug reports using 3rd party firmware will be accepted.

giddyhup, 2024-03-11:
I am using 0.98.3 and Weather Station 1.5. In the regular Sub-GHz scanner app I can hear some sensors' bursts. One of the sensors nearby is recognized by rtl_433 as a Nexus-TH. In the Weather Station app I can see the peak bar moving but it no longer shows any found stations.

skotopes, 2024-03-11:
Was it working differently in previous official releases?

giddyhup, 2024-03-11:
Yes, I remember catching at least one or two stations. I am trying all modulations.
This is what is happening around me:
[image]

giddyhup, 2024-03-11:
I know, I just want to show that there are sensors nearby.

skotopes, 2024-03-11:
@giddyhup the question is can flipper antenna and transceiver get enough signal to demod and decode payload. 
Have you found if old firmware/app version behaves differently?

giddyhup, 2024-03-11:
OK. I'll see if I can downgrade and install an earlier app version.