flipperdevices/flipperzero-good-faps #152: NFC / RFID Detector - HID ProxPoint does not activate LF indicator

imply3289, 2024-02-25:
I'm currently experiencing an issue with the FlipperZero running the NFC / RFID Detector App.

The device does not recognise the field emitted by multiple HID ProxPoint Plus readers. I have tried on a series of different readers at the same site. None activate the LF Indicator. Doing an LF Emulation of a credential works on these readers and opens doors when the correct credential is used.
I've tried a friend's FlipperZero on the same installation, he would be one firmware version behind me. His symptoms were identical.

As a troubleshooting step, I used a Proxmark3 RDV4 to create an LF field using the HID reader command and the LF field indicator displayed correctly on the FlipperZero, as expected.

I apologise I don't really know much about troubleshooting issues like this on the FlipperZero, so I'm more than happy to be guided through obtaining any logs or other details required for troubleshooting.
I am be able to obtain the appropriate permission to remove a reader from the wall and provide more information about the actual reader itself should that be required.

Flipper Firmware 0.98.3
Application Version 1.2

doomwastaken, 2024-03-12:
We have 2 of those available and fully setup. Going to verify it and get back to you

doomwastaken, 2024-03-15:
So far we were not able to reproduce it, maybe we have readers to close to one another, today we will disassemble part of our bench and try on a separate reader.

imply3289, 2024-03-15:
@doomwastaken I have ongoing access to the readers that I experience this issue with. I also have access to a proxmark3 if there's any more data I can collect about the reader?