flipperdevices/flipperzero-good-faps #203: Mass Storage not visible on Linux

yasbean, 2024-04-15:
I have tried to use Mass Storage to initiate and read on Debian Bookworm but it is not visible when plugged in to USB.  It works fine on Windows 11, though.  I am using Official FW 100.3.

skotopes, 2024-04-26:
Can you attach kernel logs?

yasbean, 2024-04-26:
Yes, here is the relevant part of /var/log/kernel.log:

`2024-04-26T10:15:28.751624-07:00 SacBean2 kernel: [  196.368983] Code: c0 84 c0 66 0f 1f 44 00 00 0f 85 71 fe ff ff 48 8b bc 24 80 00 00 00 e9 55 fe ff ff 0f 1f 44 00 00 48 8b 7b 60 31 f6 48 8b 07 <ff> 50 68 48 8b 73 60 48 8b 7b 38 41 b9 05 00 00 00 c7 04 24 21 00
2024-04-26T11:02:28.479605-07:00 SacBean2 kernel: [ 3016.093228] usb 1-1.2: new full-speed USB device number 14 using xhci_hcd
2024-04-26T11:02:28.579520-07:00 SacBean2 kernel: [ 3016.196456] usb 1-1.2: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 1.00
2024-04-26T11:02:28.579540-07:00 SacBean2 kernel: [ 3016.196464] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
2024-04-26T11:02:28.579542-07:00 SacBean2 kernel: [ 3016.196467] usb 1-1.2: Product: Flipper Kamu
2024-04-26T11:02:28.579543-07:00 SacBean2 kernel: [ 3016.196470] usb 1-1.2: Manufacturer: Flipper Devices Inc.
2024-04-26T11:02:28.579545-07:00 SacBean2 kernel: [ 3016.196472] usb 1-1.2: SerialNumber: 64 MB
2024-04-26T11:02:28.635555-07:00 SacBean2 kernel: [ 3016.252245] usb-storage 1-1.2:1.0: USB Mass Storage device detected
2024-04-26T11:02:28.635586-07:00 SacBean2 kernel: [ 3016.252669] scsi host3: usb-storage 1-1.2:1.0
2024-04-26T11:02:28.635591-07:00 SacBean2 kernel: [ 3016.253029] usbcore: registered new interface driver usb-storage
2024-04-26T11:02:28.643540-07:00 SacBean2 kernel: [ 3016.259111] usbcore: registered new interface driver uas
2024-04-26T11:02:29.663646-07:00 SacBean2 kernel: [ 3017.278065] scsi 3:0:0:0: Direct-Access     Flipper  Mass Storage     0001 PQ: 0 ANSI: 4
2024-04-26T11:02:29.663683-07:00 SacBean2 kernel: [ 3017.278853] sd 3:0:0:0: Attached scsi generic sg1 type 0
2024-04-26T11:02:29.667555-07:00 SacBean2 kernel: [ 3017.282289] sd 3:0:0:0: [sdb] 131072 512-byte logical blocks: (67.1 MB/64.0 MiB)
2024-04-26T11:02:29.667581-07:00 SacBean2 kernel: [ 3017.284620] sd 3:0:0:0: [sdb] Write Protect is off
2024-04-26T11:02:29.667585-07:00 SacBean2 kernel: [ 3017.284630] sd 3:0:0:0: [sdb] Mode Sense: 03 00 00 00
2024-04-26T11:02:29.671595-07:00 SacBean2 kernel: [ 3017.286912] sd 3:0:0:0: [sdb] No Caching mode page found
2024-04-26T11:02:29.671622-07:00 SacBean2 kernel: [ 3017.286923] sd 3:0:0:0: [sdb] Assuming drive cache: write through
2024-04-26T11:02:29.691585-07:00 SacBean2 kernel: [ 3017.308104]  sdb:
2024-04-26T11:02:29.691596-07:00 SacBean2 kernel: [ 3017.308434] sd 3:0:0:0: [sdb] Attached SCSI removable disk
2024-04-26T11:02:46.851667-07:00 SacBean2 kernel: [ 3034.467456] sd 3:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=17s
2024-04-26T11:02:46.851697-07:00 SacBean2 kernel: [ 3034.467471] sd 3:0:0:0: [sdb] tag#0 Sense Key : Hardware Error [current] 
2024-04-26T11:02:46.851700-07:00 SacBean2 kernel: [ 3034.467477] sd 3:0:0:0: [sdb] tag#0 Add. Sense: No additional sense information
2024-04-26T11:02:46.851701-07:00 SacBean2 kernel: [ 3034.467484] sd 3:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 01 ff 80 00 00 08 00
2024-04-26T11:02:46.851703-07:00 SacBean2 kernel: [ 3034.467488] I/O error, dev sdb, sector 130944 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
2024-04-26T11:02:47.011642-07:00 SacBean2 kernel: [ 3034.625277] sd 3:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=0s
2024-04-26T11:02:47.011674-07:00 SacBean2 kernel: [ 3034.625291] sd 3:0:0:0: [sdb] tag#0 Sense Key : Hardware Error [current] 
2024-04-26T11:02:47.011677-07:00 SacBean2 kernel: [ 3034.625297] sd 3:0:0:0: [sdb] tag#0 Add. Sense: No additional sense information
2024-04-26T11:02:47.011681-07:00 SacBean2 kernel: [ 3034.625304] sd 3:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 01 ff 80 00 00 08 00
2024-04-26T11:02:47.011683-07:00 SacBean2 kernel: [ 3034.625308] I/O error, dev sdb, sector 130944 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
2024-04-26T11:02:47.011684-07:00 SacBean2 kernel: [ 3034.625325] Buffer I/O error on dev sdb, logical block 16368, async page read
`

yasbean, 2024-10-14:
Any news here?  This would be very helpful on Linux.

skotopes, 2024-10-14:
@yasbean task is in backlog