flipperdevices/flipperzero-good-faps #209: [NFC Magic] User dictionary is not beeing used

M-wie-Moehre, 2024-05-02:
I have a Mifare Classic 1K key fob where some sectors require a specific key that is in the user dictionary.
When reading the key fob with the normal nfc app, everything works and all sectors are read.
However when using nfc magic, only the first few sectors (not with the specific key) are read successfully and the all other sectors aren't.
For the sectors, that aren't read immediately, it goes through the system dictionary but apparently can't find the key.
Do I have to enable the user dictionary for nfc magic or am I missing something else?

M-wie-Moehre, 2024-05-06:
I now figured out, that nfc magic really doesn't use the user dictionary. When downloading the system dict and manually adding the specific key (from the user dict) to it, it works fine. The only problem with that is, that the system dict will be overwritten, when upgrading the firmware.

Astrrra, 2024-05-13:
Can you please provide the full dump of the card you're writing to and the dump you're trying to write to it, as well as your user dictionary? I'm not able to reproduce your issue right now, the user dictionary is used correctly in my testing.

M-wie-Moehre, 2024-05-13:
Sure, here are the files. Since it's a demostration, I've just put some random numbers in block 8 that I want to write.
[files.zip](https://github.com/flipperdevices/flipperzero-good-faps/files/15298713/files.zip)

Astrrra, 2024-06-19:
Please check that the issue has been solved and re-open if not

closed 2024-06-19 by Astrrra