flipperdevices/flipperzero-good-faps #213: [NFC Magic] Issues with gen4 magic card acting as amiibo tags

smolbun, 2024-06-03:
I've wiped the gen4 magic card before writing my amiibo tag, the amiibo tag has all the pages read and is able to be detected by the Switch as an amiibo when the f0 emulates it.

When authing the gen4 magic card using the f0, not all pages can be unlock despite using "auth as ameebo".
I compared the 2 files as shown:
Left original
[image]
[image]

Astrrra, 2024-07-29:
Hi! This seems to be a problem with how Gen4 cards emulate NTAG215, with the issue being that they seem to take the PACK value (usually stored in page 134) from page 19, which contains regular amiibo data (and that data can be seen in page 134 of the resulting dump). 

Because the Gen4 cards aren't documented too well, there currently is no known way to reconfigure the Gen4 card to use the proper PACK location in NTAG215 mode. If you are able to use some other tool to write the dump to the Gen4 card and have it work as an amiibo properly - please let us know, that will help us fix that issue.

Astrrra, 2024-08-01:
Closing this issue since we are unable to fix this on our side, please re-open if you know a way to write amiibo data to a Gen4 card and have it work with the consoles.

closed 2024-08-01 by Astrrra