flipperdevices/flipperzero-good-faps #230: Support for TCM Thermo Sensors

TheDoctor200, 2024-06-23:
Hi there, I would like to receive signals from my Tchibo Weather Stataion, is this possible? 

Model: 220063 TCM

skotopes, 2024-06-23:
@TheDoctor200 we need some raw recordings of the weather station signals

TheDoctor200, 2024-06-23:
Can I do this via the flipper zero?

skotopes, 2024-06-23:
Yes, do the subghz read raw. 

@Skorpionm can you give our fellow detailed instructions ?

Skorpionm, 2024-06-24:
Use a frequency analyzer to find out what frequency the data is transmitted at. Run read raw, set the frequency, and record 10+ messages from the temperature sensor and take a photo of what the receiver shows. Also record several messages with negative temperatures.

TheDoctor200, 2024-06-24:
[image]

Some Frequency's from the Frequency analysis

TheDoctor200, 2024-06-24:
[image]

That's what reading raw gave me

Skorpionm, 2024-06-25:
To begin with, it is highly advisable to use the official firmware. You need to record raw at a frequency of 433.92. and make several records with negative and positive temperatures with photographs of the receiving part of the weather station, so that you can see what data the sensor actually transmits

TheDoctor200, 2024-06-29:
Alright I am on vacation next week expect the screenshots the week after