flipperdevices/flipperzero-good-faps #237: T5577 Feature Request - dump / restore / compare feature

amalg, 2024-07-31:
The low frequency T5577 chip is so versatile that even access control system manufacturers are using it over purchasing AWiD, EM, HID Prox, etc. LF chipsets, and replacement fob merchants are using them now too. Sometimes there are additional application data written to the T5577 and used by these systems than just their analog front-end configurations and ID data. This makes cloning the EM ID from a source T5577 to a fresh T5577 ineffective.

A good feature would be to be able to perform a complete memory dump of a T5577 chip to the Flipper Zero, then be able to write this complete memory dump to a new T5577. This would include block 0 config data and any additional application data, thus making it a perfect clone. Additionally, it would be really nice to be able to compare a dump file to a tag to check / ensure the write process has completed successfully.

skotopes, 2024-07-31:
@amalg there are physical limitations in LF-RFID subsystem that prevents us from implementing Reader-Talk-First protocols reliably. Because of that PR with HiTag support stuck in limbo.

Exactly same thing happens with 5577: implementing proper data exchange is nearly impossible.

skotopes, 2024-07-31:
Also wrong repo, should be in firmware.