flipperdevices/flipperzero-good-faps #238: Flipper crashed and was rebooted  furi_check failed

MBudkin, 2024-08-08:
Flipper crashed and was rebooted
furi_check failed

NFC Magic gen2

[image]

[image]
[image]
[image]

skotopes, 2024-08-08:
Please install official firmware and redo tests

MBudkin, 2024-08-09:
Thanks, reinstallation solved the problem.

closed 2024-08-09 by MBudkin