flipperdevices/flipperzero-good-faps #246: Weather Station - Add support for La Crosse Technology Wireless Temperature Sensor TX37-IT

LittleHelper00, 2024-10-01:
### Describe the enhancement you're suggesting.

Could you please add the support to the **Weather Station** app for **La Crosse Technology** Wireless Temperature Sensor **TX37-IT** ?
- Transmission Frequency: **868.40 MHz**
- Protocol: La Crosse Technology **IT+** _(Instant Transmission)_
- Website: https://www.lacrossetechnology.com/products/tx37u-it

Thank you 😉

skotopes, 2024-10-01:
@LittleHelper00 can you tell us device FCC-ID ? usually it's on the back of the device

LittleHelper00, 2024-10-02:
@skotopes I’ve checked the product but unfortunately there is no FCC ID present on it. I also searched on the FCC website, but it’s not listed there either.

[image]

This wireless sensor is part of the [Weather Station WS9135](https://en.lacrossetechnology.fr/P-6-A1-WS9135.html), but even after checking both the product and the FCC website, there’s no mention of its FCC ID.
[image]

Based on similar products, it should have an FCC ID similar to "OMO-TX37," but that specific FCC ID doesn’t seem to exist among the ones [registered by La Crosse Technology](https://fcc.report/company/La-Crosse-Technology-L-T-D).

The [user manual](https://en.lacrossetechnology.fr/telechargement/manuels/en/WS9135_EN.pdf) of the Weather Station WS9135 seems interesting, but it doesn't provide much information about the protocol.

However, I’ve noticed that useful information about the adopted protocol is available at this [webpage](http://fredboboss.free.fr/articles/tx29.php/) and also in this [issue](https://github.com/merbanan/rtl_433/issues/477).

I hope this helps not only with the TX37 model but also in expanding support for this great Flipper Zero app to other popular wireless sensors marketed by La Crosse Technology.

Thanks!

LittleHelper00, 2025-04-27:
@skotopes I have saved two raw bin files at two different sensor temperatures (please refer to each file name for details). I hope they will be useful. Thank you!

skotopes, 2025-04-27:
@hedger FYI