flipperdevices/flipperzero-good-faps #254: NFC Magic - Mifare Classic 1k keys and access bits can't be changed

M-wie-Moehre, 2024-10-16:
I wanted to change the access bits and both keys of one sector of my Mifare Classic 1k keyfob. The access bits of the sector allow the change of both keys. When trying to change the keys with nfc-magic, the flipper says that writing was sucessful. The keys haven't changed though.
I was now able to change the keys with an android app called "MIFARE Classic Tool", which means it is possible to change them.

skotopes, 2024-10-21:
@M-wie-Moehre are you sure that your fob is magic card?

M-wie-Moehre, 2024-10-24:
Yes, I was able to change the sector trailer of the sector from `FFFFFFFFFFFF FF078069 FFFFFFFFFFFF` to `FFFFFFFFFFFF 08778F69 FFFFFFFFFFFF` by using nfc magic.

The first access bits (`FF0780`) (should) use key A for authenticating the sector trailer, while the second access bits (`08778F`) (should) use key B for authentication (at least for writing the keys and access bits).

I now want to change both keys to something else, let's say `123456789123`. With my phone I was able to change the keys, but not with nfc magic.